nakujaproject/flight-control-computer #10: Add reverse voltage proctection

0x6f736f646f, 2022-01-19:
[image]

0x6f736f646f, 2022-02-01:
- Also don't you need a diode and a resistor at the gate pin?
- What about using a Schottky diode?

[image]

SammyOina, 2022-02-01:
There's a number of ways to do this and you don't have to have the diode you can see [here](https://components101.com/articles/design-guide-pmos-mosfet-for-reverse-voltage-polarity-protection).
But I'll look into the resistor bit

0x6f736f646f, 2022-02-01:
From one of the [references](https://components101.com/articles/design-guide-pmos-mosfet-for-reverse-voltage-polarity-protection) you have sent, "_The 100R resistor is the MOSFET gate resistor connected with the Zener diode. The Zener diode protects the gate from overvoltage_."

SammyOina, 2022-02-01:
The purpose is to clamp the gate-source voltage to a safe region within VGS max. In the case of the transistor, we're using the max is +- 12V and given there is an external power supply of 5V regulated, there is no reason to assume the voltage will go above 12V so as to damage the FET
you can see the datasheet [here](https://datasheet.lcsc.com/lcsc/1810171817_Alpha---Omega-Semicon-AO3401A_C15127.pdf)

0x6f736f646f, 2022-02-01:
> The purpose is to clamp the gate-source voltage to a safe region within VGS max. In the case of the transistor, we're using the max is +- 12V and given there is an external power supply of 5V [...]

Understood. Since we will be getting +5V from the avionics board. Then on the avionics PCB, we can include this

@Mike-Kimani

closed 2022-02-01 by 0x6f736f646f

Mike-Kimani, 2022-02-11:
> Understood. Since we will be getting +5V from the avionics board. Then on the avionics PCB, we can include this
>
> @Mike-Kimani

@0x6f736f646f Yes we will.