nakujaproject/flight-control-computer #5: RXD Led

0x6f736f646f, 2022-01-19:
What voltage are the RXD and TXD lines?
Would we need a resistor?

[image]

SammyOina, 2022-01-25:
I'm not sure about this one

0x6f736f646f, 2022-01-25:
@shohei and @Mike-Kimani

shohei, 2022-01-25:
What is the objective of this serial communication? Is the flight controller communicating with other micro controllers?

0x6f736f646f, 2022-01-25:
The objective of the serial communication is to be able to program the board

0x6f736f646f, 2022-01-25:
It is communicating with other micro controllers but not via serial

SammyOina, 2022-01-29:
[image]
I have tried a similar approach as in the Arduino mega
see [here](https://www.arduino.cc/en/uploads/Main/arduino-mega2560-schematic.pdf)

0x6f736f646f, 2022-01-30:
> [image] I have tried a similar approach as in the Arduino mega see [here](https://www.arduino.cc/en/uploads/Main/arduino-mega2560-schematic.pdf)

Grear 👍🏿

Mike-Kimani, 2022-01-30:
Good catch @0x6f736f646f .

closed 2022-02-01 by 0x6f736f646f