nakujaproject/flight-control-computer #6: Decoupling capacitors

0x6f736f646f, 2022-01-19:
Add decoupling capacitors to the VCC pins. Just an enhancement to smooth out the power

[image]

[image]

SammyOina, 2022-01-29:
we have some on the VR side

0x6f736f646f, 2022-01-30:
"They are typically applied to power sources to provide a localized source of instantaneous current and provide isolation of the local circuit from power noise in other areas of the design." by [post](https://www.arrow.com/en/research-and-events/articles/why-decoupling-capacitors-matter)

Those on VR aren't they bulk capacitors?

0x6f736f646f, 2022-02-01:
Might there be a reason you are using 100nF on the MPU6050 while the others are 0.1uF?
[image]

SammyOina, 2022-02-01:
simillarly for the mpu 6050

0x6f736f646f, 2022-02-01:
What about the 100nF cap on the MPU6050?

SammyOina, 2022-02-01:
what about it?

0x6f736f646f, 2022-02-01:
My suggestion was to use the same size of cap unless stated by the datasheet. In this case, MPU6050 can use 0.1uF unless stated by the datasheet. 

This will reduce the variants of components we have on the board. What do you think?

SammyOina, 2022-02-01:
Mathematically they are equal, I don't get the issue

0x6f736f646f, 2022-02-01:
Yes but they will be different as 100nF is C307331 and 0.1uF is C307416

SammyOina, 2022-02-01:
ok, I'll switch it then

0x6f736f646f, 2022-02-01:
Done by https://github.com/nakujaproject/flight-control-computer/pull/12/commits/82f3980e374aa3844d1cb4faae2ecbc800300b11

closed 2022-02-01 by 0x6f736f646f