nakujaproject/flight-control-computer #14: PCB antenna

0x6f736f646f, 2022-02-01:
# Issue

move the antenna out of the PCB surface

# Demo

[image]
[image]

# Ref
1. Comment on [LinkedIn](https://www.linkedin.com/feed/update/urn:li:activity:6860134286629068800?commentUrn=urn%3Ali%3Acomment%3A%28activity%3A6860134286629068800%2C6860348725047496704%29)

0x6f736f646f, 2022-02-01:
@shohei and @valNyerere why does the PCB antenna needs to be over air?

shohei, 2022-02-01:
@0x6f736f646f I checked the PCB design book. The wiring on the surface works as another antenna that interferes with the ESP's antenna. The GND plane even becomes an antenna if the shape has a sharp region. The via holes can also become antennae for the multilayer boards.

closed 2022-02-01 by 0x6f736f646f