nakujaproject/flight-control-computer #29: Change IMU

0x6f736f646f, 2022-03-13:
It is rather expensive @ $7.9 per piece and they are actually 4 in stock

0x6f736f646f, 2022-03-13:
Have you considered using the QMI8658C aka C2842151

## Ref
1. [Datasheet](https://datasheet.lcsc.com/lcsc/2108161930_QST-QMI8658C_C2842151.pdf)

SammyOina, 2022-03-15:
I think considering the rewriting of firmware to support this IMU will mess with the N2 timelines

SammyOina, 2022-03-15:
> It is rather expensive @ $7.9 per piece and they are actually 4 in stock

This will change

SammyOina, 2022-03-15:
maybe the ICM-42688-P can replace the MPU 6050 but I'll have a look first

0x6f736f646f, 2022-03-16:
> > It is rather expensive @ $7.9 per piece and they are actually 4 in stock
> 
> This will change

When will it change?

0x6f736f646f, 2022-03-16:
> I think considering the rewriting of firmware to support this IMU will mess with the N2 timelines

Aren't there libraries?

SammyOina, 2022-03-18:
> > I think considering the rewriting of firmware to support this IMU will mess with the N2 timelines
> 
> Aren't there libraries?

Haven't seen any

0x6f736f646f, 2022-03-18:
Either way your current MPU is out of stock

SammyOina, 2022-03-18:
> QMI8658C

still workin on it, i'm now considering LSM6DS3TR-C

0x6f736f646f, 2022-03-18:
I guess if we are using the `Wire.h` library it doesn't matter what IMU is on the other side of the i2c connection

SammyOina, 2022-03-18:
I'm connecting with SPI for the higher sampling frequency

closed 2022-03-22 by 0x6f736f646f